SarthakSKumar/IoT-Projects-and-Scripts #29: Broken Link in CONTRIBUTING.md

gregoris-io, 2022-10-28:
<h3>Broken link in the subsection:</h3>  

[image]

Following the link, I am brought to:  
[image]

<b>Is the intention to direct users to the Website repository, or the actual website? <b>

gregoris-io, 2022-10-28:
@SarthakSKumar, I would be more than happy to correct this issue for you upon further instruction.

Thanks, Mitch

SarthakSKumar, 2022-10-28:
Hello @gregoris-io . Thanks for bringing this to my notice. I want the link to redirect to the website, whose link is in the description of this repo. I shall assign this to you. Are you a hacktoberfest participant?

gregoris-io, 2022-10-28:
@SarthakSKumar , I will gladly accept the assignment, and yes I am a hacktoberfest participant.

SarthakSKumar, 2022-10-28:
Okay. You have 24 hours starting from now. Good Luck! @gregoris-io

closed 2022-10-29 by SarthakSKumar