SarthakSKumar/IoT-Projects-and-Scripts #36: HEARTBEAT Sensor using Arduino

Sohoxic, 2022-10-29:


SarthakSKumar, 2022-10-30:
Could you please elaborate on the implementation?

Sohoxic, 2022-10-30:
Photoplethysmography is a principle that measures the changes in the volume of blood in an organ. It does this by measuring the changes in the intensity of the light passing through the organ. This principle works behind the working of the Heartbeat Sensor. 

For my project I'm making use of an IR LED to be the light source in the heartbeat sensor, and photodetectors like Photo Diode, Photoresistor, or an LDR can be used and then arranging the components , i.e., the light source and photodetector, in two ways: Transmissive Sensor and Reflective Sensor

Sohoxic, 2022-10-30:
This project is an inexpensive alternative to Smart Watches and other expensive Heart Rate, Monitors.

SarthakSKumar, 2022-10-30:
You can work on the issue @Sohoxic . You have 24 hours starting from now