SarthakSKumar/IoT-Projects-and-Scripts #78: Bug: Fix footer links not working

SarthakSKumar, 2023-04-09:


Preetraj2002, 2023-10-18:
@SarthakSKumar can you specify which link is not working. I have tested the site, I see 4 links are working fine and other 2 legal links have no webpages.

SarthakSKumar, 2023-10-19:
Hi @Preetraj2002 
The existing sections need to be removed from the footer. 
New sections to be created in the footer:

'Contributing'
1. Code of Conduct: https://github.com/SarthakSKumar/IoT-Projects-and-Scripts/blob/master/CODE_OF_CONDUCT
2. Guidelines: https://github.com/SarthakSKumar/IoT-Projects-and-Scripts/blob/master/CONTRIBUTING.md

'About'
1. About Repo: https://github.com/SarthakSKumar/IoT-Projects-and-Scripts/blob/master/README.md
2. Maintainer: https://github.com/SarthakSKumar

'Projects'
1. Arduino Projects: https://github.com/SarthakSKumar/IoT-Projects-and-Scripts/tree/master/Arduino%20Projects
2. ESP Projects: https://github.com/SarthakSKumar/IoT-Projects-and-Scripts/tree/master/ESP%20Projects

The social links and icons have to be removed as well.

Preetraj2002, 2023-10-20:
I have updated the footer. Please review it.

SarthakSKumar, 2023-10-20:
Hi @Preetraj2002 
Please don't create PR's until an issue is assigned to you. Assigning now. Will review the PR

closed 2023-10-20 by SarthakSKumar

Preetraj2002, 2023-10-20:
Sure. I always do wait till I am assigned a task but few org don't formally assign so I thought you would just review it. Will make sure next time :)

PR #87 (2023-10-20): Updated the footer and removed social links
Successfully updated the footer and its links.
Removed the social links and logo. 
Fixes #78